ajstarks/openvg #22: Random crashes or random shapes start to appear

pawelduda, 2015-01-17:


pawelduda, 2015-01-19:
I think I have sort of fixed the issue. I removed all references to two other typefaces I wasn't using. After that the program was running for 20 hours without any crash. Maybe the fact that they consumed too much memory was the problem?

ajstarks, 2015-01-20:
if you use other typefaces, you have to be sure to free the resources.

pawelduda, 2015-01-20:
I am reporting this because it occurred when the 3 standard font typefaces that were defined in libshapes.c (though I added ~150 additional glyphs to each so that I could use Polish special characters - but as far as I recall, the program crashed regardless of that).

My program relies on running constantly so there is no point in freeing font resource that is being used.

Since I only used standard sans typeface, I removed the other two from libshapes.c and anywhere else. 

After removing both serif and mono typefaces, the program is still running properly after 30+ hours. Before it would always crash somewhere between 10-90 minutes. I believe the issue could be narrowed down to this. 

It could turn out to be a pitfall for people wanting to use your library and falling into same issue later on. Perhaps moving the font loading somewhere else or not including all 3 typefaces by default could avoid a lot of confusion in the future.

ajstarks, 2015-01-20:
thanks for the report.  I’ll look into the font handling.  Does your fork contain your glyph addition code?

pawelduda, 2015-01-20:
Yes, it does. The current version has 400 glyphs and one font face. But if you want the crashing version (3 font faces, 500 glyphs each), go with the first commit of my fork.

pawelduda, 2015-01-21:
I've had the issue happen again after messing with my program. Haven't done anything out of ordinary, just moved text and images around. Path around the letter "r" (for some reason it ALWAYS seems to be that one letter) got corrupted. 

sudo vcdbg reloc stats showed that there was heap corruption or pointer out of bounds (not always though). Also, GPU had only 2.4MB free memory. I have just increased memory split for GPU and I'll let you know if the issue persists.

ajstarks, 2015-01-23:
I have reproduced the issue with the digital_signage.c program (crashed after a few hours), but after recompiling to remove the warnings (removed the reference to the unused bg image, added function prototypes for Clear and CreateImageFromJpeg functions to shapes.h, the example has been running for 12 hours.

ajstarks, 2015-01-23:
My test has been running for 24 hours with no crashes.

pawelduda, 2015-01-23:
I am afraid this could be purely a coincidence as I have removed all of the warnings recently and I'm 80% sure it crashed afterwards.

Since I am not 100% sure, I will keep on testing and if it crashes without warnings, I'll make a commit.

ajstarks, 2015-01-24:
FYI: My test has run for 48 hours with no crash..

pawelduda, 2015-01-24:
I have left my program overnight and it didn't crash after more than 20h. I assumed that everything is OK.

Then I ran it again and one of glyph paths got corrupted almost instantly (after 5 seconds). There was no compilation or any data change in between. 

I suspect this issue has to do with some resources not being released properly? Because the crash is more likely to happen on a subsequent program run. I am actually not sure whether it ever happened when I just ran the program directly after booting up my Pi.

I have just commited version that crashed today.

ajstarks, 2015-01-24:
I’ll try a version of your code without your extended fonts to see if the problem persists.

ajstarks, 2015-01-26:
last report.  I stopped the run after 4 days without a crash.

pawelduda, 2015-01-29:
This clearly has to do with font handling mechanism. 

Just put this at start of while rendering loop:
unloadfont(SansTypeface.Glyphs, SansTypeface.Count);
SansTypeface = loadfont(DejaVuSans_glyphPoints, DejaVuSans_glyphPointIndices, DejaVuSans_glyphInstructions, DejaVuSans_glyphInstructionIndices, DejaVuSans_glyphInstructionCounts, DejaVuSans_glyphAdvances, DejaVuSans_characterMap, DejaVuSans_glyphCount);

You will almost instantly notice same side effects (random blinking of rectangles, random glyphs getting corrupted every now and then), regardless of using extended fonts or not. This should help with debugging. It seems like some sort of memory corruption in place where glyph path segments are stored.

I have tried tinkering with libshapes for a few hours but with no result.

ajstarks, 2015-01-30:
Does this occur with the unforked version?  Also, can you send a short program that reproduces the error?

ajstarks, 2015-01-30:
Was init() and finish() called?  These should be paired because they load/unload.

pawelduda, 2015-01-30:
In my program it looks like this:
init();
Start();
while(1) { //until keypress
    // render code     
    End();
}
finish();
return 0;

Finish is called once I press a key just before the program exits to clean up. Initially I followed your code made in particles example but it would break as well.

And yes, it did occur since the very beginning with the unforked version with early version of my program. I will work on a minimal example later.

What's strange, I've had an identical issue with this node.js OpenVG implementation which actually made me try yours:
https://github.com/luismreis/node-openvg-canvas

Both implementations seem to use similar font handling.

pawelduda, 2015-02-03:
Back, without the code but with some info.

I think I finally got it figured out. I believe that calling functions like Rect() or Circle() multiple in a loop caused this behavior. This is due to the fact that those functions create and then destroy a new path each time they're called, meaning that a lot of memory reallocation is going on. In my case it was very redundant to use those functions because most of shapes were static containers that never changed their position/size. 

I accidentally discovered this when I was building analog clock the clean OpenVG way. I put code like this inside main loop:

// begin loop
hour = vgCreatePath(VG_PATH_FORMAT_STANDARD, VG_PATH_DATATYPE_F, 1, 0, 0, 0, VG_PATH_CAPABILITY_ALL);
min = vgCreatePath(VG_PATH_FORMAT_STANDARD, VG_PATH_DATATYPE_F, 1, 0, 0, 0, VG_PATH_CAPABILITY_ALL);
hour_counter = vgCreatePath(VG_PATH_FORMAT_STANDARD, VG_PATH_DATATYPE_F, 1, 0, 0, 0, VG_PATH_CAPABILITY_ALL);
min_counter = vgCreatePath(VG_PATH_FORMAT_STANDARD, VG_PATH_DATATYPE_F, 1, 0, 0, 0, VG_PATH_CAPABILITY_ALL);
sec_counter = vgCreatePath(VG_PATH_FORMAT_STANDARD, VG_PATH_DATATYPE_F, 1, 0, 0, 0, VG_PATH_CAPABILITY_ALL);
vgAppendPathData(hour, 4, clock_cmds, hour_coords);
vgAppendPathData(min, 4, clock_cmds, min_coords);
vgAppendPathData(hour_counter, 4, clock_cmds, hour_counter_coords);
vgAppendPathData(min_counter, 4, clock_cmds, min_counter_coords);
vgAppendPathData(sec_counter, 4, clock_cmds, sec_counter_coords);

// (rest of loop, do the drawing stuff here)

vgDestroyPath(hour);
vgDestroyPath(min);
vgDestroyPath(hour_counter);
vgDestroyPath(min_counter);
vgDestroyPath(sec_counter);
// end loop

The result was something like a random font glyph getting corrupted after few seconds of runtime, then another after few more seconds. I quickly realized my mistake and fixed the code. Since then - no problem.

So I replaced all of those Rect() calls with reusable paths, created at start of the program and released them at the end. Now the only operation that is being repeated is drawing are matrix transformations and vgDrawPath() calls which is the way it should be. Not only it gave me a noticeable speed-up due to the fact that I was reusing already stored paths but also I haven't seen any path blink/mutate for 2 days of heavy tinkering.

void Rect(VGfloat x, VGfloat y, VGfloat w, VGfloat h) {
	VGPath path = newpath(); <--
	vguRect(path, x, y, w, h);
	vgDrawPath(path, VG_FILL_PATH | VG_STROKE_PATH);
	vgDestroyPath(path); <--
}

ajstarks, 2015-02-04:
Thanks for the debugging.  Do you suggest an updated API?

pawelduda, 2016-03-30:
I have had a few people reach out to me to ask whether the fix worked in the end, so I decided to post an update here just to let everyone interested know.

One year later, having my application in a production environment all the time, with at least 12h uptime a day, not a single issue like this was ever reported to me.

So in short: if you struggle with this, try applying the above fix.

ajstarks, 2016-04-15:
I'm reopening to track

ykram, 2016-06-04:
I'm glad this is reopened as I think I'm hitting something similar (redirected to this issue per https://www.raspberrypi.org/forums/viewtopic.php?f=69&t=144790)

I have an application that reads data via UDP socket and uses this library to graph & plot shapes depending on the data read in via the UDP socket. I've noticed that after a bit of runtime, typically less than an hour, the video will cease to refresh and the process which is doing the OpenVG calls has locked completely. SIGINT and SIGKILL doesn't appear to terminate the process.

Anytime I re-run the process, OpenVG calls don't appear to work.
Backtrace as per a gdb session:
```
#0  0x76d8ba40 in do_futex_wait (isem=isem@entry=0x76c29a40 <khrn_queue+76>)
    at ../nptl/sysdeps/unix/sysv/linux/sem_wait.c:48
#1  0x76d8baf4 in __new_sem_wait (sem=0x76c29a40 <khrn_queue+76>)
    at ../nptl/sysdeps/unix/sysv/linux/sem_wait.c:69
#2  0x76b51aa4 in vchiu_queue_pop () from /opt/vc/lib/libvchiq_arm.so
#3  0x76c02be8 in rpc_recv () from /opt/vc/lib/libEGL.so
#4  0x76c132dc in vguLine () from /opt/vc/lib/libEGL.so
#5  0x76da9920 in Line () from /usr/lib/libshapes.so
#6  0x43b66666 in ?? ()
```

ajstarks, 2016-06-06:
I believe there is a larger problem here: see: https://github.com/raspberrypi/linux/issues/943