ajstarks/openvg #24: Scissoring support

lobsangludd, 2015-02-09:
I added to functions to my copy:

`// LimitTo limits the drawing area to specified rectangle
void LimitTo(VGint x, VGint y, VGint w, VGint h) {
        vgSeti(VG_SCISSORING, VG_TRUE);
        VGint coords[4] = { x, y, w, h };
        vgSetiv(VG_SCISSOR_RECTS, 4, coords);
}

// Unlimit stops limiting drawing area to specified rectangle
void Unlimit() {
        vgSeti(VG_SCISSORING, VG_FALSE);
}`

If you find them useful, please add them.

ajstarks, 2015-02-09:
looks interesting.  Can you post a sample code that uses these?

lobsangludd, 2015-02-09:
I use it mostly for limiting display of scrolling text but I assume there are other uses:

int main(int argc, char **argv) {
  int w, h;
  init(&w, &h);

  const char *str = "Blockquotes are very handy in email to emulate reply text. This line is part of the same quote.";

  int i = 0;
  while (1) {
    Start(w,h);
    Background(255, 255, 255);

    LimitTo(0, 0, 1000, 200);
    {
      Fill(0, 0, 0, 1);
      Translate(i-=10,0);
      Text(0, 0, str, SansTypeface, 50);
    }
    Unlimit();

    End();
  }
}

lobsangludd, 2015-02-09:
Maybe this is a better example: draws three circles, the middle one is a semicircle:

  Start(w,h);
  Background(255, 255, 255);

  Fill(255, 0, 0, 1);
  Circle(50, 50, 100);

  LimitTo(150, 150, 100, 100);
  {
    Circle(150, 150, 100);
  }
  Unlimit();

  Circle(250, 250, 100);

  End();

I also must thank you for a great library! =)

ajstarks, 2015-02-10:
How about ClipRect and ClipEnd as function names?
Should there be a ClipPath function as well?

lobsangludd, 2015-02-10:
ClipRect and ClipEnd sounds good. Scissoring to multiple rectangles is supported by spec, but there is no path option for this function.

lobsangludd, 2015-02-10:
I could possibly be implemented with Masking function, but I'll have to test it out

ajstarks, 2015-02-11:
try this:
```
//
// clip: test rectangular clipping
//
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>
#include <string.h>
#include <fcntl.h>
#include <time.h>
#include "VG/openvg.h"
#include "VG/vgu.h"
#include "shapes.h"

int main(int argc, char **argv) {
        int w, h, fontsize;
        VGfloat x, cx, cy, cw, ch, midy, speed;
        char *message = "Now is the time for all good men to come to the aid of the party";
        char done[3];

        init(&w, &h);
        speed = 10;
        midy = (VGfloat) h / 2;
        fontsize = h / 20;
        cx = 0.0;
        ch = fontsize * 2;
        cw = w;
        cy = midy - (ch / 2);

        for (x = 0; x < cw+1; x += speed) {
                Start(w, h);
                Background(255, 255, 255);
                Fill(0,0,0,.2);
                Rect(cx, cy, cw, ch);
                ClipRect(cx, cy, cw, ch);
                Translate(x, cy + (fontsize / 2));
                Fill(0, 0, 0, 1);
                Text(0, 0, message, SansTypeface, fontsize);
                ClipEnd();
                End();
        }
        fgets(done, 2, stdin);
        finish();
        exit(0);
}
```

ajstarks, 2015-02-11:
Library has been updated with ClipRect/ClipEnd

closed 2015-02-11 by ajstarks