ajstarks/openvg #27: unwanted echo of keystrokes to  terminal

ottoecheverri, 2015-06-13:
In running a openvg program, all keystrokes as well as mouse buttons are echo to whatever was open before, the terminal or xterm. In my openvg program I use raw keystrokes using 
open("/dev/input/event0",O_RDONLY|O_NONBLOCK)
I don't want the echo, any sugestions?
Thank you.

ajstarks, 2015-06-13:
try the  standard functions: saveterm(), rawterm(), restoreterm().

closed 2015-10-28 by ajstarks

ottoecheverri, 2016-04-07:
Thank you for your answer, sorry I didn't reply sooner.
I'm still getting echo into the xterminal from the mouse and keyboard.
I suppose is because I don't know the proper attributes for the rawterm() function.
I use "open("/dev/input/event0".O_RDWR | O_NONBLOCK)" to get the raw key codes with "mouse0" for the mouse codes.
I appreciate your help very much.
Also, can the SaveEnd function or something else be used to create a JPG or BMP screenshot of a VG  screen?