ajstarks/openvg #28: after installation: don't work, lib not found

VogonJeltz, 2015-10-24:
after installation: don't work, lib not found

source code: your helloWorld example
~~~
openvg_helloworld.c:6:23: fatal error: VG/openvg.h: no such file or directory
#include "VG/openvg.h"
^
compilation terminated.
~~~

which are the parameters which have to be passed go g++ call additionionally?
unfortunately your description is misleading.
~~~
currently my function call by Geany settings is
for compile:
g++ -Wall -c "%f" -lpthread  -lrt -lwiringPi
for make:
g++ -Wall -o "%e" "%f" -lpthread -lpigpio -lrt -lwiringPi 
~~~

what has to be added for your graph libs?

VogonJeltz, 2015-10-24:
what I did so far:

pi@raspberrypi ~ $ sudo apt-get install libjpeg8-dev indent libfreetype6-dev ttf-dejavu-core

Next, build the library and test:

pi@raspberrypi ~ $ git clone git://github.com/ajstarks/openvg
pi@raspberrypi ~ $ cd openvg
pi@raspberrypi ~/openvg $ make
g++ -I/usr/include/freetype2 fontutil/font2openvg.cpp -o font2openvg -lfreetype
./font2openvg /usr/share/fonts/truetype/ttf-dejavu/DejaVuSans.ttf DejaVuSans.inc DejaVuSans
224 glyphs written
./font2openvg /usr/share/fonts/truetype/ttf-dejavu/DejaVuSansMono.ttf DejaVuSansMono.inc DejaVuSansMono
224 glyphs written
./font2openvg /usr/share/fonts/truetype/ttf-dejavu/DejaVuSerif.ttf DejaVuSerif.inc DejaVuSerif
224 glyphs written
gcc -O2 -Wall -I/opt/vc/include -I/opt/vc/include/interface/vmcs_host/linux -I/opt/vc/include/interface/vcos/pthreads -c libshapes.c
gcc -O2 -Wall -I/opt/vc/include -I/opt/vc/include/interface/vmcs_host/linux -I/opt/vc/include/interface/vcos/pthreads -c oglinit.c
pi@raspberrypi ~/openvg/client $ cd client
pi@raspberrypi ~/openvg/client $ make test
cc -Wall -I/opt/vc/include -I/opt/vc/include/interface/vcos/pthreads -o shapedemo shapedemo.c ../libshapes.o ../oglinit.o -L/opt/vc/lib -lGLESv2 -ljpeg
./shapedemo demo 5

pi@raspberrypi ~/openvg/client $ cd ~/openvg
pi@raspberrypi ~/openvg $ make library
gcc -O2 -Wall -I/opt/vc/include -I/opt/vc/include/interface/vmcs_host/linux -I/opt/vc/include/interface/vcos/pthreads -c libshapes.c
gcc -L/opt/vc/lib -lGLESv2 -lEGL -ljpeg -shared -o libshapes.so oglinit.o libshapes.o
pi@raspberrypi ~/openvg $ sudo make install
install -m 755 -p font2openvg /usr/bin/
install -m 755 -p libshapes.so /usr/lib/libshapes.so.1.0.0
strip --strip-unneeded /usr/lib/libshapes.so.1.0.0
ln -f -s /usr/lib/libshapes.so.1.0.0 /usr/lib/libshapes.so
ln -f -s /usr/lib/libshapes.so.1.0.0 /usr/lib/libshapes.so.1
ln -f -s /usr/lib/libshapes.so.1.0.0 /usr/lib/libshapes.so.1.0
install -m 644 -p shapes.h /usr/include/
install -m 644 -p fontinfo.h /usr/include/

VogonJeltz, 2015-10-24:
But:
what is this??
pi@raspberrypi ~/openvg $ gcc -I/opt/vc/include anysource.c -o anysource -lshapes

how has this to be changed to work with Geany?
???

Ebiroll, 2015-10-24:
You probably have to configure geany  Build -> Set build commands and options (this might differ from version of Geany you are using)

This should give you a hint.  gcc   -I/opt/vc/include -o anysource.c -lshapes

Probably you want to add -lshapes and  -I/opt/vc/include to  get something like this
g++ -Wall -o "%e" "%f" -lpthread  -I/opt/vc/include  -lrt -lwiringPi  -lshapes
  
Note -I (capital i)  -I/opt/vc/include
Note -l (capital l)   -lshapes

VogonJeltz, 2015-10-24:
I'll copy and paste it, perhapd all I tries was by lowercase l(L)
a better font in this forum would be appreciated ;)

now it would read like this:

~~~
for compile:
g++ -Wall -c "%f" -lpthread  -lrt -lwiringPi  -I/opt/vc/include -lshapes
for make:
g++ -Wall -o "%e" "%f" -lpthread  -lrt -lwiringPi -I/opt/vc/include -lshapes
~~~
correct ?

VogonJeltz, 2015-10-24:
no, again lots of errors!

g++ -Wall -o "openvg_helloworld.c" -lpthread -lrt -lwiringPi -I/opt/vc/include -lshapes (im Verzeichnis: /home/pi/programs/openvg)
/usr/lib/gcc/arm-linux-gnueabihf/4.9/../../../arm-linux-gnueabihf/crt1.o: In function `_start':
/build/glibc-IRmfEu/glibc-2.19/csu/../ports/sysdeps/arm/start.S:119: undefined reference to `main'
collect2: error: ld returned 1 exit status
Kompilierung fehlgeschlagen.

Ebiroll, 2015-10-24:
Try compiling on the command line.
g++ -Wall -o test.exe  openvg_helloworld.c -lpthread -lrt -lwiringPi -I/opt/vc/include -lshapes

VogonJeltz, 2015-10-24:
command line for your vghello in client subdir worked

whats wrong with 
~~~
In function `_start':
/build/glibc-IRmfEu/glibc-2.19/csu/../ports/sysdeps/arm/start.S:119: undefined reference to `main'
~~~

Ebiroll, 2015-10-24:
You have no main() function  :-P

VogonJeltz, 2015-10-24:
it's your original code!

VogonJeltz, 2015-10-24:
// first OpenVG program
// Anthony Starks (ajstarks@gmail.com)
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>
#include "VG/openvg.h"
#include "VG/vgu.h"
#include "fontinfo.h"
#include "shapes.h"

int main() {
    int width, height;
    char s[3];

    init(&width, &height);                  // Graphics initialization

    Start(width, height);                   // Start the picture
    Background(0, 0, 0);                    // Black background
    Fill(44, 77, 232, 1);                   // Big blue marble
    Circle(width / 2, 0, width);            // The "world"
    Fill(255, 255, 255, 1);                 // White text
    TextMid(width / 2, height / 2, "hello, world", SerifTypeface, width / 10);  // Greetings
    End();                                  // End the picture

    fgets(s, 2, stdin);                     // look at the pic, end with [RETURN]
    finish();                               // Graphics cleanup
    exit(0);
}

VogonJeltz, 2015-10-24:
again:

pi@raspberrypi ~/openvg/client $ g++ -Wall -o test.exe hellovg.c -lpthread -lrt -lwiringPi -I/opt/vc/include -lshapeshell
hellovg.c: In function ‘int main()’:
hellovg.c:22:77: warning: deprecated conversion from string constant to ‘char*’ [-Wwrite-strings]
TextMid(width / 2, height / 2, "hello, world", SerifTypeface, width / 10); // Greetings
^
/usr/bin/ld: cannot find -lshapeshell
collect2: error: ld returned 1 exit status
Nach oben

VogonJeltz, 2015-10-24:
isn't it usually
int main(int argc, char *argv[]) ?

Ebiroll, 2015-10-24:
Try
g++ -Wall -o test.exe hellovg.c -lpthread -lrt -lwiringPi -I/opt/vc/include -lshape

VogonJeltz, 2015-10-24:
~~~
pi@raspberrypi ~/openvg/client $ g++ -Wall -o test.exe hellovg.c -lpthread -lrt -lwiringPi -I/opt/vc/include -lshape
hellovg.c: In function ‘int main()’:
hellovg.c:22:77: warning: deprecated conversion from string constant to ‘char*’ [-Wwrite-strings]
TextMid(width / 2, height / 2, "hello, world", SerifTypeface, width / 10); // Greetings
^
/usr/bin/ld: cannot find -lshape
collect2: error: ld returned 1 exit status
pi@raspberrypi ~/openvg/client $ g++ -Wall -o test.exe hellovg.c -lpthread -lrt -lwiringPi -I/opt/vc/include -lshape
hellovg.c: In function ‘int main()’:
hellovg.c:22:77: warning: deprecated conversion from string constant to ‘char*’ [-Wwrite-strings]
TextMid(width / 2, height / 2, "hello, world", SerifTypeface, width / 10); // Greetings
^
/usr/bin/ld: cannot find -lshape
collect2: error: ld returned 1 exit status
~~~

VogonJeltz, 2015-10-24:
I suggest you try to fix this issue, my Raspi has no internet connection any more and its effing lame!

Ebiroll, 2015-10-24:
OK. But you need to write -lshapes not -lshape

VogonJeltz, 2015-10-24:
aaah, a typo - write it again, just moment....

all those annoying vogon poetry to write each time anew... :(

VogonJeltz, 2015-10-24:
yes, just a warning about conversion from *char to string !

what is this test.exe thing ???

VogonJeltz, 2015-10-24:
program cannot be stopped by a enter key stroke!

VogonJeltz, 2015-10-24:
again, if I compile my own file 
openvg_helloworld.c

again it does not work.

F9:
...(a moment...)

VogonJeltz, 2015-10-24:
please fix this!
I'm using raspbian. latest build,
and geany, latest build.

ajstarks, 2015-10-25:
The issue is that all of the include paths are not accounted for for a standalone build.
I will fix.

In the meantime, as a work around, if you move your code into the client directory, and model the build as in the Makefile, it will work.

ajstarks, 2015-10-25:
The paths are correct. you must specify these input paths:

 gcc -Wall -I/opt/vc/include -I/opt/vc/include/interface/vmcs_host/linux -I/opt/vc/include/interface/vcos/pthreads mycode.c -o mycode-lshapes

I have updated the README

VogonJeltz, 2015-10-25:
thank you!
so the Geany paths have to be all in one...
~~~
// link/make:
g++ -Wall -o test.exe "%e" "%f" -lpthread -lrt -lwiringPi -I/opt/vc/include -I/opt/vc/include/interface/vmcs_host/linux -I/opt/vc/include/interface/vcos/pthreads mycode.c -o mycode-lshapes
~~~
?
is that correct?

Ebiroll, 2015-10-25:
Hello. I dont have geany installed. I will try in next week.
The option -o test.exe 
Should generate the file test.exe for running.
It should probably read -o  "%e" to generate the name of your projectfile but somehow "%e" was empty.

VogonJeltz, 2015-10-25:
sorry, I'm still too much a beginner - I don't understand yet.
Can I drop the -o test.exe thing then finally?

can you pleas give me a COMPLETE parameter line how it has to look like at the end of one's tether
(not singular improvement notices, please) ?

VogonJeltz, 2015-10-25:
by 

(for compile:) g++ -Wall -o  "%f" -lpthread -lrt -lwiringPi -I/opt/vc/include -I/opt/vc/include/interface/vmcs_host/linux -I/opt/vc/include/interface/vcos/pthreads  -o -lshapes

(for make:) g++ -Wall -o "%e" "%f" -lpthread -lrt -lwiringPi -I/opt/vc/include -I/opt/vc/include/interface/vmcs_host/linux -I/opt/vc/include/interface/vcos/pthreads  -o -lshapes

I now get:
g++ -Wall -o  "openvg_helloworld.c" -lpthread -lrt -lwiringPi -I/opt/vc/include -I/opt/vc/include/interface/vmcs_host/linux -I/opt/vc/include/interface/vcos/pthreads  -o -lshapes (im Verzeichnis: /home/pi/programs/openvg)
Kompilierung fehlgeschlagen.
/usr/lib/gcc/arm-linux-gnueabihf/4.9/../../../arm-linux-gnueabihf/crt1.o: In function `_start':
/build/glibc-IRmfEu/glibc-2.19/csu/../ports/sysdeps/arm/start.S:119: undefined reference to `main'
collect2: error: ld returned 1 exit status

ajstarks, 2015-10-25:
your command line is incorrect, use this one: (given code in mysource.c)

 ```
$  gcc -I/opt/vc/include -I/opt/vc/include/interface/vmcs_host/linux -I/opt/vc/include/interface/vcos/pthreads mysource.c -o anysource -lshapes
$ ./mysource
```

Here is the explanation:

1) gcc (the command the compile your code)
2) -I/opt/vc/include -I/opt/vc/include/interface/vmcs_host/linux -I/opt/vc/include/interface/vcos/pthreads  (the directories to search for include files)
3) mysource.c  (your file that you want to build)
4) -o mysource  (the output of the executable, this is what you run)
5)  -lshapes (the OpenVG library that you installed)
6) ./mysource (run the executable you just built)

VogonJeltz, 2015-10-25:
I'm using g++, not gcc
 how is the parameter line fusioned with

(for compile:) g++ -Wall -o "%f" -lpthread -lrt -lwiringPi 

(for make:) g++ -Wall -o "%e" "%f" -lpthread -lrt -lwiringPi
?

I don't use command line ever, far too much to type every time, I am only using the Geany parameters

ajstarks, 2015-10-25:
anysource.c is a "placeholder" for your code. FYI: see https://www.raspberrypi.org/forums/viewtopic.php?f=69&t=28990

VogonJeltz, 2015-10-25:
g++ -Wall -I/opt/vc/include -I/opt/vc/include/interface/vmcs_host/linux -I/opt/vc/include/interface/vcos/pthreads "%e"-o "%f" -lshapes -lpthread -lrt -lwiringPi

?

VogonJeltz, 2015-10-25:
g++ -Wall -I/opt/vc/include -I/opt/vc/include/interface/vmcs_host/linux -I/opt/vc/include/interface/vcos/pthreads "hellovg"-o "hellovg.c" -lshapes -lpthread -lrt -lwiringPi (im Verzeichnis: /home/pi/openvg/client)
Kompilierung fehlgeschlagen.
g++: error: hellovg-o: Datei oder Verzeichnis nicht gefunden

ajstarks, 2015-10-25:
the -o option is incorrect.  it should be -o hellovg

VogonJeltz, 2015-10-25:
I have no hellovg in my Genay parameters, just "%e" and "%f"

VogonJeltz, 2015-10-25:
so this way...?

g++ -Wall -I/opt/vc/include -I/opt/vc/include/interface/vmcs_host/linux -I/opt/vc/include/interface/vcos/pthreads -o "%e" "%f" -lshapes -lpthread -lrt -lwiringPi

VogonJeltz, 2015-10-25:
I think I have it:
Geany settings for compile:
g++ -Wall -I/opt/vc/include -I/opt/vc/include/interface/vmcs_host/linux -I/opt/vc/include/interface/vcos/pthreads -c  "%f" -lshapes -lpthread -lrt -lwiringPi

Geany settings for make:
g++ -Wall -I/opt/vc/include -I/opt/vc/include/interface/vmcs_host/linux -I/opt/vc/include/interface/vcos/pthreads -o "%e" "%f" -lshapes -lpthread -lrt -lwiringPi

seems to work, just a warning about strings:
hellovg.c:22:77: warning: deprecated conversion from string constant to ‘char*’ [-Wwrite-strings]

closed 2015-10-25 by VogonJeltz