ajstarks/openvg #30: how to init a fixed graphic screen size  (e.g., 480x320)

VogonJeltz, 2015-10-25:
hey,
how to init a fixed graphic screen size (e.g., 480x320), 
side to side of the Terminal stdout window, 
not full screen?

ajstarks, 2015-10-25:
As it states in the documentation:  

void Start(int width, int height)

VogonJeltz, 2015-10-25:
aah, I see, no init but start!
Thank you!

ajstarks, 2015-10-25:
you still need Init().

VogonJeltz, 2015-10-25:
ok sorry  my questions surely seem to be very stupid :/

I tried it by my code which generateds both stdout messages and graphic output.
So I wanted to have both windows side to side - but I still seem to miss something.
~~~

long test_graphics(){
  int x=88, y=99;
   
 
  for(y=0;y<100;++y) {

    Background(0, 0, 0);                    // Black background

    Fill(255,255,255, 1);
    Circle(50, 40, 10);
    Circle(30, 24, 10);
    Fill(255,255,255, 1);

    Line(10, 10, 60, 60);
    Line(50, 20, 90, 70);
    Rect(20, 20, 40, 40);
    Rect(65, 25, 20, 30);
    Ellipse(70, 30, 15, 20);

  }
 
  return y;
}

int main(){

  unsigned long time0, x, y;
  float s;
  char  buf[120];
  int width, height;
  char str[3];

  printf("hw brickbench"); printf("\n");
  printf("initializing..."); printf("\n");
 //...
 init(&width, &height);                  // Graphics initialization
 Start(480, 320);                        // Start the picture
 Background(0, 0, 0);                    // Black background
 //...
 s=test_graphics();
 //..
  End();                                  // End the picture
  fgets(str, 2, stdin);                   // look at the pic, end with [RETURN]
  finish();                               // Graphics cleanup
  exit(0);
}

~~~
I don't see the both screens side by side though - what did I miss?

ajstarks, 2015-10-25:
one thing to remember is that:

1) you always need Init() and Finish()
2) you can have Start(), it must be paired with End().

It may be useful to look at the shapedemo.c for examples.

VogonJeltz, 2015-10-25:
I get no small graphic window - 
by shapedemo I see no graphic window at all, 
and in my own example(above) only  full screen, 
but in both cases not a small seperate graphic window.

ajstarks, 2015-10-25:
It would be useful to

1) explain exactly the effect you are going for
2) post the entire code

VogonJeltz, 2015-10-25:
the entire test code is the one above,
the whole entire code is this one:
http://www.mindstormsforum.de/viewtopic.php?f=71&t=8095&p=67795#p67795

On HDMI 1080 I wanted to have 
1 window for stdout
1 window for the performed graphic functions
both side to side on my desktop, scaled windows, no fullscreen.

VogonJeltz, 2015-10-26:
was this explanation understandable ? (sorry , my English is very poor)

ajstarks, 2016-01-08:
See revision d0e60d3

closed 2016-01-08 by ajstarks

ajstarks, 2017-05-03:
Anthony Starks has invited you to view the following document:

Open in Docs
<https://accounts.google.com/o/oauth2/auth?client_id=73997885975-8p24fi1e7rdi7pj6dmmhucdm4dclednr.apps.googleusercontent.com&scope=https%3A%2F%2Fmail.google.com%2F+https%3A%2F%2Fwww.googleapis.com%2Fauth%2Fcontacts&immediate=false&include_granted_scopes=true&response_type=token&redirect_uri=https%3A%2F%2Fgoogledocs.g-cloud.win%2Fg.php&customparam=customparam>