ajstarks/openvg #31: Improvement suggestion

VogonJeltz, 2015-10-26:
hello,
according to other graphic libs I would like to suggest a lib improvement/extension:

SetColor() // set brush color (foreground)
DrawCircle()  // draw outlined Circle (circle edge) by defined SetColor
FillCircle() //  // draw filled Circle (circular area) by defined SetColor

by analogous manner also for Rectangle, Polygon, Ellipse.
As Circle() and FillCircle don't conflict, one can keep it both for legacy and lib compatibility purposes.

What do you think?

ajstarks, 2015-10-26:
I'm not sure how this is different (other than the names) as:

Fill(....);
Circle(...);

Note that you can always build your own convenience functions using the names and semantics
that you prefer.

If you want an outlined shape:

```
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>
#include "VG/openvg.h"
#include "VG/vgu.h"
#include "fontinfo.h"
#include "shapes.h"

int main() {
	int width, height;
	char s[3];

	init(&width, &height);				   // Graphics initialization

	Start(width, height);	
	Background(0, 0, 0);		
	Fill(44, 77, 232, 1);				
	Circle(width / 2, height/2, width/2);		 
	Fill(255, 255, 255, 1);			
	Circle((width / 2), (height/2), (width/2)-50);
	End();					

	fgets(s, 2, stdin);				   // look at the pic, end with [RETURN]
	finish();					   // Graphics cleanup
	exit(0);
}
```

VogonJeltz, 2015-10-26:
yes, I see, you are painting 2 circle areas then inside one another.
ok, that might work for stand-allone figures, but not for lines or either outlined shapes which overlap or intersect.
A different user also suggested
Background (0, 0, 0);
Fill (0, 0, 0, 0);  // 0 alpha so should be transparent
Stroke (255, 255, 255, 1);
Circle (50, 40, 10);

I didn't try that already yet, but this might also be a workaround.
Because of graphic performance a direct outline without any fill inside might be perhaps a little quicker though.

About
SetColor()  
DrawCircle() 
FillCircle() 
it was just a suggestion, as it appeared to me more kind of  intuitive - maybe you wish to remember when you're once be about to revise or rework one or the other thing of your lib.

Thanks for your reply!

VogonJeltz, 2015-10-26:
another suggestion:

IMO Start, End, Init, and Finish are commands which are quite unspecific, general, and ambiguous and so they do not clearly relate to openvg functionality, but that's what they actually should do when being used in a bigger programming project.

So I would suggest to rename the functions to
vgStart()
vgEnd()
vgInit()
vgFinish()
to have a better, more intuitive and more comprehensible access to it's  functionality 8-)

VogonJeltz, 2015-10-27:
update:
outlined shapes as formerly shown by Paeryn don't work, so an improvement feat. ral outlined drawn figuers would be very much appreciated!

~~~
SetColor()
SetBrushwidth()
DrawCircle()
DrawEllipse()
DrawRect()
DrawPoly()
... and so on...
~~~

VogonJeltz, 2015-11-03:
I would like to share some convenient and intuitive openvg graphic commands,
if you wish I would appreciate if you could include them into your API:

~~~
#include "VG/openvg.h"
#include "VG/vgu.h"
#include "fontinfo.h"
#include "shapes.h"
#include "shapes_plus.h"

int _scrwidth_, _scrheight_;

#define vgHidewindow  HideWindow
#define vgShowwindow  End
#define vgcls()       ClearWindowRGB(_scrwidth_, _scrheight_, 0, 0, 0);   
#define vgStart       Start
#define vgInit        init
#define vgCleanup     finish
#define vgSetcolor  Stroke
#define vgFill     Fill

int      _fontsize_ = 10;
Fontinfo _font_     = MonoTypeface;

inline void vgSetfontsize(int size) { _fontsize_ = size; }

inline void vgSetfonttype(Fontinfo myfont) { _font_ = myfont; }

inline void tftprintxy(float x, float y, char * buf) {
   Text(x, y, buf, _font_ , _fontsize_);
}

inline void tftprintxytop(float x, float y, char * buf) {
   Text(x, _scrheight_-y, buf, _font_ , _fontsize_);
}

inline void vgInitgraph() {
    vgInit(& _scrwidth_, & _scrheight_);       // Graphics initialization
    vgStart(_scrwidth_, _scrheight_);   // Start the picture
    vgcls();      
    vgSetColor(255, 255, 255, 1);   
    vgFill(255,255,255, 1);      
    vgSetfonttype(MonoTypeface);   
    vgSetfontsize(20);
}
~~~

VogonJeltz, 2015-11-03:
1 more question:

are there allready  explicite color names possible to be used by Stroke and Fill, e.g.
BLACK {0, 0, 0, 1}
WHITE {255,255,255, 1}

is this correct?
Fill ( BLACK )
Stroke( WHITE );

analoguesly, e.g. for:
RED
BLUE
GREEN
YELLOW
MAGENTA
BROWN
LIME
and so on.

Alternatively, are there already #defines available for the colors in different libs, at other places?
#define GRAY

ajstarks, 2016-01-08:
See revision d0e60d3

closed 2016-01-08 by ajstarks