ajstarks/openvg #32: [type: improvement] add library request: shapes_plus by Paeryn

VogonJeltz, 2015-10-28:
add library request: shapes_plus by Paeryn https://www.raspberrypi.org/forums/memberlist.php?mode=viewprofile&u=4009
- feat. outlined shapes
- feat easier hide / show graphic canvas screen  alternating to Console/Terminal window

https://www.raspberrypi.org/forums/viewtopic.php?f=33&t=124183#p835188

https://www.dropbox.com/s/h37po1q305eab17/shapes_plus.tar.gz?dl=0

ajstarks, 2015-10-29:
Do you or Paeryn care to make a pull request?

VogonJeltz, 2015-10-29:
I personally don't understand much of this lib and how everything works - but perhaps Paeryn will, he seems to understand much about your lib.

As he said, maybe he will make a "fork", too (not sure actually what this actually means).

ajstarks, 2016-01-08:
See revision d0e60d3

closed 2016-01-08 by ajstarks