ajstarks/openvg #35: [improvement suggestion] color codes for standardized color names

VogonJeltz, 2015-11-03:
if you wish you are welcome to include the following color codes for standardized color names
(source: http://www.december.com/html/spec/colordeccompact.html) - additional wishes and proposals appreciated!
~~~
    #define RED      255, 0, 0, 1
    #define DEEPPINK 255, 20, 147, 1
    #define ROSE     255, 0, 204, 1
    #define MAGENTA  238, 0, 238, 1
    #define FUCHSIA  255, 0, 255, 1
    #define YELLOW   255, 255, 0, 1
    #define SIGNYELLOW 255, 209, 22, 1
    #define ORANGE   255, 102, 0, 1
    #define LIME     0, 255, 0, 1
    #define GREEN3   0, 205, 0. 1
    #define DARKGREEN   47, 79, 47, 1
    #define CYAN     0, 255, 255, 1
    #define BLUE     0,  0,  255, 1
    #define AQUA     0, 255, 255, 1
    #define BROWN   128, 42, 42, 1
    #define SIGNBROWN   96, 51, 17, 1
    #define OCHRE   204, 119, 34, 1
    #define GRAY25   64, 64, 64, 1
    #define GRAY50  127, 127, 127, 0
    #define GRAY75  191, 191, 191, 1
    #define BLACK    0,  0,  0,  1
~~~

VogonJeltz, 2015-11-11:
I updated the color names and color code constants just to be integrated into your libs, without extra #include needed if possible, to be used by the end-user right out of the box.
HTH !

ajstarks, 2016-01-08:
See revision d0e60d3

closed 2016-01-08 by ajstarks