ajstarks/openvg #42: Updated font handing

ajstarks, 2016-01-09:
This is a tracking bug to discuss the implementation of updated font handling.

TaoHawaii, 2017-10-21:
TextMid() doesn't correctly set the Y offset.   I am making digital gauges and I used "double radius*sin(pi/180)+(width/2)" to get the Y coordinate of evenly spaced text around the center of the screen.  I used the same code (copy and paste) to draw small circles around the points where the TextMid() characters should be.  The center of the circles lines up very well with the bottom of the characters all the way around.

paeryn, 2017-10-21:
That is how it is supposed to be, the text will be centre justified at the point given.

Text() draws the string left justified,
TextMid() draws the string centre justified,
TextEnd() draw the string right justified.

In all three the baseline of the font will be on the point. Were you thinking that the centre of the bounding box of the text would be at that point?

TaoHawaii, 2017-10-22:
Yes, that;s what I was thinking.  Thanks!