ajstarks/openvg #45: Very slow on rpi2 raspbian

raph-amiard, 2016-01-24:
Running this simple example:

~~~c
// first OpenVG program
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>
#include "shapes.h"
#include <time.h>

#include <sys/time.h>

long long current_timestamp() {
    struct timeval te;
    gettimeofday(&te, NULL);
    long long milliseconds = te.tv_sec*1000LL + te.tv_usec/1000;
    return milliseconds;
}

int main() {
    long long start, end;
    int width, height;
    VGfloat w2, h2, w;
    char s[3];

    init(&width, &height);                                      // Graphics initialization

    w2 = (VGfloat)(width/2);
    h2 = (VGfloat)(height/2);
    w  = (VGfloat)w;
    int circ_size = 0;
    int i;

    while (1) {
        circ_size = (circ_size + 1) % width;
        start = current_timestamp();
        Start(width, height);                                       // Start the picture
        Background(0, 0, 0);                                        // Black background
        Fill(44, 77, 232, 1);                                       // Big blue marble
        Circle(w2, 0, circ_size);                                           // The "world"
        Fill(255, 255, 255, 1);                                     // White text
        TextMid(w2, h2, "hello, world", SerifTypeface, width/10);   // Greetings
        End();                                                      // End the picture
        end = current_timestamp();
        printf("Loop iteration: %dms\n", end - start);
    }

    finish();                                                   // Graphics cleanup
    exit(0);
}
~~~

Compiled with

~~~
gcc vgtest.c libshapes.c oglinit.c -I../src -I/opt/vc/include/ -I../example -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -L/opt/vc/lib -lEGL -lm -lGLESv2 -lbcm_host -ljpeg -o test.bin
~~~

Yields a constant 338ms for the iteration time on a raspberry pi 2. This seems incredibly slow for hardware accelerated vector graphics. Are those results known ? Am I doing something wrong ?

ajstarks, 2016-01-26:
What is acceptable speed of rendering is dependent on your app, but here is an alternative benchmark.
Note that I moved invariants out of the loop.

```
//  OpenVG benchmark
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>
#include <shapes.h>
#include <time.h>

#include <sys/time.h>

int main() {

    int width, height;
    VGfloat w2, h2, w;
    char s[3];

    init(&width, &height);                                      // Graphics initialization

    w2 = (VGfloat)(width/2);
    h2 = (VGfloat)(height/2);
    w  = (VGfloat)w;
    int circ_size = 0;
    int i;

    Start(width, height);                                       // Start the picture
    Background(0, 0, 0);                                        // Black background
    Fill(255, 255, 255, 1);                                     // White text
    TextMid(w2, h2, "hello, world", SerifTypeface, width/10);   // Greetings
    Fill(44, 77, 232, 1);                                       // Big blue marble
    int iterations = 1000;
    struct timeval start, stop;
    double elapsed = 0.0;
    while (iterations--) {
        circ_size = (circ_size + 1) % width;
        gettimeofday(&start, NULL);
        Circle(w2, 0, circ_size);                              // The "world"
        End();
        gettimeofday(&stop, NULL);
        elapsed = (double)(stop.tv_usec - start.tv_usec) / 1000000 + (double)(stop.tv_sec - start.tv_sec);
        printf("Elaspsed = %.6f\n", elapsed);
    }
    finish();                                                 // Graphics cleanup
    exit(0);
}
```

raph-amiard, 2016-01-26:
Hello Anthony, thanks for your answer !

And sorry also, my example was wrong the printf should have read:

~~~C
printf("Loop iteration: %lldms\n", end - start);
~~~

With this corrected printf, the iteration time in my benchmark is actually equivalent as the one in yours, so around ~= 17ms. Actually, this simple example: 

~~~
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>
#include "shapes.h"
#include <sys/time.h>

long long current_timestamp() {
    struct timeval te;
    gettimeofday(&te, NULL);
    long long milliseconds = te.tv_sec*1000LL + te.tv_usec/1000;
    return milliseconds;
}

int main() {
    long long start, end;
    int width, height;
    init(&width, &height);                                      // Graphics initialization
    Start(width, height);                                       // Start the picture
    while (1) {
        start = current_timestamp();
        End();                                                      // End the picture
        end = current_timestamp();
        printf("Loop iteration: %lldms\n", end - start);
    }

    finish();                                                   // Graphics cleanup
    exit(0);
}
~~~

Also caps at 17ms per iteration, which gives a framerate of 58fps which is the screen's refresh rate, so I think the opengl swap buffers caps the framerate on the screen's refresh rate.

closed 2016-01-26 by ajstarks