ajstarks/openvg #50: compiling library error

steverpi, 2016-02-27:
When I 'cd' into the openvg folder and 'make library' I get the following mo rule error

$ make library
gcc -O2 -Wall -I/opt/vc/include -I/opt/vc/include/interface/vmcs_host/linux -I/opt/vc/include/interface/vcos/pthreads -fPIC -c oglinit.c
make: *** No rule to make target '/usr/share/fonts/truetype/ttf-dejavu/DejaVuSans.ttf', needed by 'DejaVuSans.inc'.  Stop.

The library does not get made.  I have not altered anything in the freshly downloaded openvg folder.

I did not get this error before and it used to make fine, however I was using RPi Wheezy, but now I have just done a fresh install of Jessie OS.  Is there anything else needed to do a make library?

Thanks
Steve

ajstarks, 2016-02-27:
Jessie changed the location of the TTF files. 
see the FONTLIB variable in the Makefile.

ajstarks, 2016-02-27:
documentation updated by revision b43a87c

closed 2016-02-27 by ajstarks

steverpi, 2016-02-27:
Wow, that was quick!  Thanks again for all the excellent updates - it is a superb library.  Now back to sorting out a small (odd shaped, not rect) mouse cursor routine which does not get through to the desktop behind. If I manage it, I will let you know.

 

About saving screenshots in jpg, png, etc, is this something that I could mention on the forum.  The reason I ask is because it is not a fault, it is a library enhancement.