ajstarks/openvg #53: Enabling swapInterval?

ykram, 2016-05-19:
Apologies if this has is documented somewhere but as I couldn't find it I figured I'd raise an issue regarding enablement of vsync.

Is there anyway to enable vsync for apps this library? Right now I'm set to call eglSwapInterval in oglinit() but I'm curious if there's a way to programatically do it without having to modify, recompile and re-link to the modified library.

paeryn, 2016-05-19:
Swapinterval is 1 by default and if you need any other value you need to modify oglinit as you said since the display handle isn't exposed outside of the library. 
Any value other than 0 should force the screen update to by vsynced, so it should already be enabled.

ykram, 2016-05-20:
Where is the call to the vsync call if you don't mind me asking? I looked
but couldn't find it called anywhere.

paeryn, 2016-05-20:
Vsync automatically happens when the swapinterval is > 0. There is no call to eglSwapInterval at the moment as it defaults to 1 anyway (which is by far the most common setting). There is no separate call to enable or disable vsync, if you want it enabled set a swapinterval > 0 (which is how it is set up) , to disable it set the swapinterval to 0.

ykram, 2016-05-20:
Ah, got it. Thanks for pointing that out.

closed 2016-05-20 by ykram