ajstarks/openvg #54: function dot(x,y) or pixel(x,y missing) ?

tofrnr, 2016-06-12:
hello,
I can't find a function to set a point or a dot or a pixel at (x,y)
is it really missing?

ajstarks, 2016-06-12:
Note that this is a vector library, and pixel-level operations are usually not done at this level.
A possible workaround is to draw a rectangle or circle with the desired dimensions.  

What is your application doing?

tofrnr, 2016-06-12:
it's putting a series of function values 
f(x) = y 
in a coordinate system, shown as (x,y)

ajstarks, 2016-06-12:
I suggest, picking the right scale for your app, and use circles to represent the points.  See: https://github.com/ajstarks/openvg/blob/master/go-client/vgplot/vgplot.go for an example.

tofrnr, 2016-06-12:
yes, but that's too much computation IMO

ajstarks, 2016-06-12:
if you know the (x,y) coordinates, just place an appropriately sized circle.

tofrnr, 2016-06-12:
yes, I understood your proposal, but as I said: too much computation to do.
There are about 10000-40000 points to draw each second.

ajstarks, 2016-06-12:
With that throughput requirement you will need to pre-allocate the object and reuse using low-level OpenVG calls.

tofrnr, 2016-06-12:
That's supposed to be too hard to understand to me, I just wanted to put single pixels by simple pixel(x,y) API calls.

ajstarks, 2016-06-12:
There is a fork of the library that pre-allocates: https://github.com/paeryn/openvg/tree/newfonts

tofrnr, 2016-06-12:
thank you, But there I also can't find a pixel(x,y) API function:

~~~
extern void Translate(VGfloat, VGfloat);
	extern void Rotate(VGfloat);
	extern void Shear(VGfloat, VGfloat);
	extern void Scale(VGfloat, VGfloat);
	extern void Text(VGfloat, VGfloat, const char *, Fontinfo, int);
	extern void TextMid(VGfloat, VGfloat, const char *, Fontinfo, int);
	extern void TextEnd(VGfloat, VGfloat, const char *, Fontinfo, int);
	extern VGfloat TextWidth(const char *, Fontinfo, int);
	extern void Cbezier(VGfloat, VGfloat, VGfloat, VGfloat, VGfloat, VGfloat, VGfloat, VGfloat);
	extern void Qbezier(VGfloat, VGfloat, VGfloat, VGfloat, VGfloat, VGfloat);
	extern void Polygon(VGfloat *, VGfloat *, VGint);
	extern void Polyline(VGfloat *, VGfloat *, VGint);
	extern void Rect(VGfloat, VGfloat, VGfloat, VGfloat);
	extern void Line(VGfloat, VGfloat, VGfloat, VGfloat);
	extern void Roundrect(VGfloat, VGfloat, VGfloat, VGfloat, VGfloat, VGfloat);
	extern void Ellipse(VGfloat, VGfloat, VGfloat, VGfloat);
	extern void Circle(VGfloat, VGfloat, VGfloat);
	extern void Arc(VGfloat, VGfloat, VGfloat, VGfloat, VGfloat, VGfloat);
	extern void Image(VGfloat, VGfloat, int, int, char *);
	extern void Start(int, int);
	extern void End();
	extern void SaveEnd(char *);
	extern void Background(unsigned int, unsigned int, unsigned int);
	extern void BackgroundRGB(unsigned int, unsigned int, unsigned int, VGfloat);
	extern void init(int *, int *);
	extern void finish();
	extern void setfill(VGfloat[4]);
	extern void setstroke(VGfloat[4]);
	extern void StrokeWidth(VGfloat);
	extern void Stroke(unsigned int, unsigned int, unsigned int, VGfloat);
	extern void Fill(unsigned int, unsigned int, unsigned int, VGfloat);
	extern void RGBA(unsigned int, unsigned int, unsigned int, VGfloat, VGfloat[4]);
	extern void RGB(unsigned int, unsigned int, unsigned int, VGfloat[4]);
	extern void FillLinearGradient(VGfloat, VGfloat, VGfloat, VGfloat, VGfloat *, int);
	extern void FillRadialGradient(VGfloat, VGfloat, VGfloat, VGfloat, VGfloat, VGfloat *, int);
	extern void ClipRect(VGint x, VGint y, VGint w, VGint h);
	extern void ClipEnd();
	extern Fontinfo loadfont(const int *, const int *, const unsigned char *, const int *, const int *, const int *,
				 const short *, int, int, int);
	extern void unloadfont(Fontinfo);
	extern void makeimage(VGfloat, VGfloat, int, int, VGubyte *);
	extern void saveterm();
	extern void restoreterm();
extern void rawterm();
~~~

ajstarks, 2016-06-12:
As I said, substitute pixel(x,y), with Circle(x,y,size), where size is appropriately scaled to your canvas. 
you could also use a Line(x, y, x+(unit), y) where unit the the appropriate size.  

You will have to determine if the throughput is sufficient for your app.

tofrnr, 2016-06-12:
yes, thank you, I assumed cit was easier to have a dedicated pixel(x,y) API function nevertheless

closed 2016-06-13 by ajstarks