ajstarks/openvg #55: Frozzen screen with the examples with alt-tab or mouse click

tarribas, 2016-08-23:
Hi, 

In my raspberry X11 terminal, when I launch any sample code like ./particles and I click with the mouse (out-side the terminal window), The focus is lost then I can't send the Ctrl-C signal and the control over the graphic is lost. 
Then I've to connect by network to my raspberry and kill the ./particles process to close the graphics window and then I can see all my keystrokes in other window.

Is normal or I'm doing anything wrong?

Regards.
Tomás.

paeryn, 2016-08-24:
None of the example programs are X11 aware. Anything you do with the mouse will be seen and interpreted by the desktop, this includes giving focus to another window.

The easiest way to get back is to put your X11 terminal in the bottom-left corner of the screen before you run the program, that way if you do change the window focus it's very easy to re-position the cursor back within the terminal's window and left-click to regain focus even though you aren't able to see the desktop (just move the mouse left-down a lot and then slightly up-right).

n2ballroy, 2017-01-14:
I am having the same problem when I ran any of the example programs such as the hello world program.  If I don't touch the mouse the hello world program returns as it should when <enter> is hit.

If I touch the mouse while the graphics window is open, then it never closes, and I have to pull power to recover.

I am using a Rpi 3 with Raspbian Jessie and a typical HDMI monitor.

The terminal program I am launching from is the standard one "LXTerminal 0.2.0" which launches from the Raspbian Jessie distro.

P.S.: I am having a similar issue with the "simple2D" libraries.