ajstarks/openvg #62: Golang make failed

thhan, 2018-02-28:
I installed all dependecies `sudo apt-get install libjpeg8-dev indent libfreetype6-dev ttf-dejavu-core`

If I start `make` in openvg I get following message:

```
gcc -O2 -Wall -I/opt/vc/include -I/opt/vc/include/interface/vmcs_host/linux -I/opt/vc/include/interface/vcos/pthreads -fPIC -c libshapes.c
gcc -L/opt/vc/lib -lEGL -lGLESv2 -ljpeg -shared -o libshapes.so oglinit.o libshapes.o
/usr/bin/ld: cannot find -lEGL
/usr/bin/ld: cannot find -lGLESv2
collect2: error: ld returned 1 exit status
Makefile:35: recipe for target 'library' failed
make: *** [library] Error 1

```

paeryn, 2018-03-02:
I take it you are on Raspbian Stretch. The names of those two libraries changed (now prefixed with **brcm**) so as to not conflict with the mesa ones (Jessie had both the new and old names for the libraries). You need to edit the Makefile to change line 2 from

`LIBFLAGS=-L/opt/vc/lib -lEGL -lGLESv2 -ljpeg`
to
`LIBFLAGS=-L/opt/vc/lib -lbrcmEGL -lbrcmGLESv2 -ljpeg`

ajstarks, 2018-03-04:
This is a recurring bug.  Will fix.