ajstarks/openvg #64: Degree sign doesn't get rendered

BrainStone, 2018-03-27:
Whenever I try to render text containing the degree sign (`°`), everything after it isn't rendered anymore. What could be the cause of that?

steverpi, 2018-03-27:
Could be same as when using micro symbol (mu) µ which is not 8-bit encoding.  May work if you set localisation - just a suggestion:

#include <locale.h>      // For use with multi-byte parsing routines (extended char sets other than 7-bit ascii) otherwise
                         // will return an empty string if you pass a string with a char val > 127
  setlocale(LC_ALL, "") ; // Set locale for everything

BrainStone, 2018-03-27:
Hm. I already set an specify locale so special chars get rendered correctly. I'll see if I can get it working. But that's a great point to start.

ajstarks, 2018-03-27:
That's odd.  Are you using the C library or Go wrapper?

I have a specific app that uses the degree symbol, twh (https://github.com/ajstarks/openvg/blob/master/go-client/twh/twh.go)

BrainStone, 2018-03-27:
I'm using the C library in a C++ project.

wasalm, 2018-04-14:
Turns out that the degree sign is not part of the Dejavusans font.
You need to find a font which contains the degree sign and turn it into c++ code with font2openvg.
If you are interested, I already compiled some weather symbols into c++.