ajstarks/openvg #68: OpenVG Crash with recent firmwares...

davidecat, 2018-10-26:
Hi, im testing the clip.c script inside a for(;;)...

```
//
// clip: test rectangular clipping
//
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>
#include "VG/openvg.h"
#include "VG/vgu.h"
#include "shapes.h"

int main() {
	int  w, h, cx, cy, cw, ch, midy, fontsize;
	char *message = "Now is the time for all good men to come to the aid of the party";
	char s[3];

	init(&w, &h);
	float speed, x, rx, ry, rw, rh;
	speed = 5.0;
	x  = 0;
	midy = (h / 2);
	fontsize = w / 50;
	cx = 0;
	ch = fontsize * 2;
	cw = w;
	cy = midy - (ch / 2);

	rx = (float)cx;
	ry = (float)cy;
	rw = (float)cw;
	rh = (float)ch;

for (;;) {

	// scroll the text, only in the clipping rectangle
	for (x = 0.0; x < rw+speed; x += speed) {
		Start(w, h);
		Background(255, 255, 255);
		Fill(0, 0, 0, .2);
		Rect(rx, ry, rw, rh);
		ClipRect(cx, cy, cw, ch);
		Translate(x, ry+(fontsize/2));
		Fill(0, 0, 0, 1);
		Text(0, 0, message, SansTypeface, fontsize);
		ClipEnd();
		End();
	}
 }
	fgets(s, 2, stdin);
	finish();
	exit(0);
}
```
With recent firmwares (i tried with 4.14.x and 4.17.x), running it on top of a fullHD video with omxplayer, it freezes often, blocking all the openVG, it needs to reboot to start again...
With the firmware 4.4.x it doesn happen, it can run for days without any issue, even together with omxplayer and fullHD videos... 
Is there any known similar issue with the new firmwares?
Thank you for your help

ajstarks, 2018-10-26:
Odd. I have not touched that code in some time.

davidecat, 2018-10-26:
yes, I guessed it's weird but there is no any other explanation... :-)
4.4.33 date is 20 Nov 2016, 2 years ago... maybe other more recent firmwares work, i did not test all... 
You can easily recreate the issue, run clip.c with a fullscreen omxplayer fullHD video...
Any suggestion?
Thank you for your help an and by the way thank you and congratulations for your great job.

ajstarks, 2018-10-27:
just rebuild clip.c on 4.14.78-v7+.  No issues.

davidecat, 2018-10-28:
@ajstarks Have u rebuild this version with the for(;;) loop? This one i have attached in my first message.
I just tried again, RPI3B, raspbian Stretch, 4.14.78-v7+, this clip.c (with the loop), omxplayer with 1 fullHD video (around 6minutes long), it crashes randomly, sometimes after 20 minutes, sometimes after 1 hour or more...
With 4.4.33 on the same RPI3B it works continously...
Thank you again for your help.

ajstarks, 2018-10-28:
FYI: Running your looping example with the latest kernel (4.14.78-v7), (without the omxplayer) for about 1 hour.

davidecat, 2018-10-28:
Thank you @ajstarks.
It runs perfectly without omxplayer... If u try with omxplayer and a fullHD video, it will become unstable..

ktb92677, 2019-07-26:
Hey are there any updates on this issue? Surely OpenVG should be able to run along side a fullHD video

closed 2024-07-18 by ajstarks