ajstarks/openvg #69: PI4 support

ktb92677, 2019-07-26:
Hey! Will this library be supported on the raspberry PI 4? This thread here seems to suggest that it might not be: https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=245380

ajstarks, 2019-07-26:
I saw this thread as well.  I'll have to get a Pi4 to test.

paeryn, 2019-07-26:
The RPi 4 doesn't support OpenVG, it uses Mesa for controlling the GPU and OpenVG was removed from Mesa years ago (through little use and lack of anybody maintaining it I think).

ajstarks, 2019-08-03:
FYI, the code will build on the Pi4, but running the executable fails with this message:

```* failed to add service - already in use?```

peternewman, 2020-01-13:
If you get this error

	* failed to add service - already in use?

You'll need to comment out the following line with a hash:

	dtoverlay=vc4-fkms-v3d

But it then hangs on a futex (with strace):
```
futex(0xb6e4c518, FUTEX_WAKE_PRIVATE, 1) = 1
ioctl(3, _IOC(_IOC_READ|_IOC_WRITE, 0xc4, 0x2, 0x1c), 0xbef8b35c) = 0
ioctl(3, _IOC(_IOC_READ|_IOC_WRITE, 0xc4, 0x2, 0x1c), 0xbef8b35c) = 0
ioctl(3, _IOC(_IOC_NONE, 0xc4, 0xd, 0), 0x44015) = 0
ioctl(3, _IOC(_IOC_NONE, 0xc4, 0xd, 0), 0x45016) = 0
mmap2(NULL, 8392704, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS|MAP_STACK, -1, 0) = 0xb49fd000
mprotect(0xb49fe000, 8388608, PROT_READ|PROT_WRITE) = 0
clone(child_stack=0xb51fcf98, flags=CLONE_VM|CLONE_FS|CLONE_FILES|CLONE_SIGHAND|CLONE_THREAD|CLONE_SYSVSEM|CLONE_SETTLS|CLONE_PARENT_SETTID|CLONE_CHILD_CLEARTID, parent_tidptr=0xb51fd4c8, tls=0xb51fd920, child_tidptr=0xb51fd4c8) = 1062
ioctl(3, _IOC(_IOC_NONE, 0xc4, 0xc, 0), 0x3f010) = 0
ioctl(3, _IOC(_IOC_NONE, 0xc4, 0xc, 0), 0x3f010) = 0
ioctl(3, _IOC(_IOC_WRITE, 0xc4, 0x4, 0xc), 0xbef8b3fc) = 0
ioctl(3, _IOC(_IOC_NONE, 0xc4, 0xd, 0), 0x3f010) = 0
ioctl(3, _IOC(_IOC_NONE, 0xc4, 0xc, 0), 0x3f010) = 0
ioctl(3, _IOC(_IOC_READ|_IOC_WRITE, 0xc4, 0x8, 0x10), 0xbef8b3d8) = 13
ioctl(3, _IOC(_IOC_NONE, 0xc4, 0xd, 0), 0x3f010) = 0
ioctl(3, _IOC(_IOC_NONE, 0xc4, 0xd, 0), 0x3f010) = 0
ioctl(3, _IOC(_IOC_READ|_IOC_WRITE, 0xc4, 0x2, 0x1c), 0xbef8b3d4) = 0
ioctl(3, _IOC(_IOC_READ|_IOC_WRITE, 0xc4, 0x2, 0x1c), 0xbef8b3d4) = 0
ioctl(3, _IOC(_IOC_READ|_IOC_WRITE, 0xc4, 0x2, 0x1c), 0xbef8b3d4) = 0
ioctl(3, _IOC(_IOC_NONE, 0xc4, 0x9, 0), 0x47018) = 1058
ioctl(3, _IOC(_IOC_WRITE, 0xc4, 0x4, 0xc), 0xbef8b3bc) = 0
ioctl(3, _IOC(_IOC_NONE, 0xc4, 0x9, 0), 0x47018) = 1058
futex(0xb6eaca58, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 1, NULL, FUTEX_BITSET_MATCH_ANY
```

peternewman, 2020-01-13:
> I've got your library working on a standard x86 Ubuntu 16.04 machine via MESA (rather messily currently, so that might be a way to make it work everywhere, although possibly not if they've dropped it).

So I've just tried running the code I tweaked on a Pi and after a little more minor tweaking, it compiles and displays, but it looks like each pixel is much bigger than it should be, for example particles is all squares, shapedemo looks similarly distorted.

ajstarks, 2020-01-13:
Thanks for trying this.   Can you post details of your tweaks?

peternewman, 2020-01-15:
Hi @ajstarks ,

I'm using https://github.com/simonhyde/PiClock/ so I thought I might have to learn about git submodules, but I realised as I got your examples working first, I could just push that so here it is:
https://github.com/peternewman/openvg/tree/crossplatform

Apologies it's quite so hacky, I've not really tidied it up, this version works what I assume is fine (I get circular particles for example on Ubuntu 16.04 with the following MESA related packages installed:
```
ii  libegl1-mesa:amd64                          18.0.5-0ubuntu0~16.04.1                         amd64        free implementation of the EGL API -- runtime
ii  libegl1-mesa-dev:amd64                      18.0.5-0ubuntu0~16.04.1                         amd64        free implementation of the EGL API -- development files
ii  libegl1-mesa-drivers:amd64                  12.0.6-0ubuntu0.16.04.1                         amd64        transitional dummy package
ii  libgl1-mesa-dev:amd64                       18.0.5-0ubuntu0~16.04.1                         amd64        free implementation of the OpenGL API -- GLX development files
ii  libgl1-mesa-dri:amd64                       18.0.5-0ubuntu0~16.04.1                         amd64        free implementation of the OpenGL API -- DRI modules
ii  libgl1-mesa-glx:amd64                       18.0.5-0ubuntu0~16.04.1                         amd64        free implementation of the OpenGL API -- GLX runtime
ii  libglapi-mesa:amd64                         18.0.5-0ubuntu0~16.04.1                         amd64        free implementation of the GL API -- shared library
ii  libgles2-mesa:amd64                         18.0.5-0ubuntu0~16.04.1                         amd64        free implementation of the OpenGL|ES 2.x API -- runtime
ii  libgles2-mesa-dev:amd64                     18.0.5-0ubuntu0~16.04.1                         amd64        free implementation of the OpenGL|ES 2.x API -- development files
ii  libglu1-mesa:amd64                          9.0.0-2.1                                       amd64        Mesa OpenGL utility library (GLU)
ii  libglu1-mesa-dev:amd64                      9.0.0-2.1                                       amd64        Mesa OpenGL utility library -- development files
ii  libopenvg1-mesa-dev-lts-utopic:amd64        3:13                                            amd64        Transitional package for libopenvg1-mesa-dev-lts-utopic
ii  libopenvg1-mesa-lts-utopic:amd64            3:13                                            amd64        Transitional package for libopenvg1-mesa-lts-utopic
ii  libtxc-dxtn-s2tc0:amd64                     0~git20131104-1.1                               amd64        Texture compression library for Mesa
ii  libwayland-egl1-mesa:amd64                  18.0.5-0ubuntu0~16.04.1                         amd64        implementation of the Wayland EGL platform -- runtime
ii  mesa-common-dev:amd64                       18.0.5-0ubuntu0~16.04.1                         amd64        Developer documentation for Mesa
ii  mesa-utils                                  8.3.0-1                                         amd64        Miscellaneous Mesa GL utilities
ii  mesa-vdpau-drivers:amd64                    18.0.5-0ubuntu0~16.04.1                         amd64        Mesa VDPAU video acceleration drivers
```

peternewman, 2020-01-20:
Now with a minor change so it compiles, but doesn't work, on a Pi 4. Reading around it sounds like it could be because I haven't turned on the stencil buffer, inspired by this:
https://github.com/memononen/nanovg#rendering-is-wrong-what-to-do

See also a lot of the code in here:
https://github.com/FlightGear/flightgear/blob/8e6500ae47ff74539cd2c0a9c4eebe96e69ca81d/utils/fgpanel/GLES_utils.cxx#L262

simonhyde, 2020-07-08:
I got it working on ShivaVG and GLUT (like the ShivaVG examples). This did mean changing the way the library worked to being driven by a callback loop, which was fine for my requirements, but obviously might not work for other applications. https://github.com/simonhyde/openvg